udacity/fcnd-issue-reports #220: motion_planning.py msgpack error

GuidoManfredi, 2018-04-02:


GuidoManfredi, 2018-04-02:
When trying to run motion_planning.py, I get the following error:
"ModuleNotFoundError: No module name 'msgpack'.

I am using ubuntu 17.10
FCND env is activated

mkontz, 2018-04-04:
I received the same error in Windows.

ryan-keenan, 2018-04-07:
Thanks for reporting this, `msgpack` was a fairly late addition to the Python starterkit.  Can you double check that it exists in the `environment.yml` file you installed? If not you'll need to update your FCND env to include it.  I'll assume this is solved but please feel free to reopen the ticket if not!

ryan-keenan, 2018-04-07:
`environment.yml` should look like this: https://github.com/udacity/FCND-Term1-Starter-Kit/blob/master/environment.yml

closed 2018-04-07 by ryan-keenan