udacity/fcnd-issue-reports #240: Drone Integration Crazyflie

mehmetyldz87, 2018-04-14:


mehmetyldz87, 2018-04-14:
when I run the backyard project code as described on the drone integration part , I hit the ceiling. Also one of the user has same issue.  

P.S: I successfully run the example project of crazyflie (  https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/motion_commander_demo.py )

adrnp, 2018-04-17:
In 'takeoff_transition()' you hqve the takeoff altitude set to 3.0 meters.  Try setting it to a smaller value (e.g. 0.5m) to make sure it stays much lower to the ground.

mehmetyldz87, 2018-04-17:
@adrnp  Upsss : ) , I got it , but I don't understand that part, how it can go up above 1 meter in hover mode

jtuckerdunn, 2018-04-20:
@adrnp can we add a sentence in the drone integration content that says to make sure you change the default byf waypoint heights if you are inside?