udacity/fcnd-issue-reports #243: index-error in A-Star-City-Solution

ywiyogo, 2018-04-15:
In the 11-th step, it throws an index error. The variable `pp` only has 1 dimensional array.
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-11-1733def4d5ac> in <module>()
      7 
      8 pp = np.array(path)
----> 9 plt.plot(pp[:, 1], pp[:, 0], 'g')
     10 
     11 plt.xlabel('EAST')

IndexError: too many indices for array

domluna, 2018-04-21:
@ywiyogo are you sure a path to the goal was found? The only case this should be 1D is in an empty array is returned.

ywiyogo, 2018-04-22:
yes, the 10th step prints this:
```
Found a path.
1023 1023.0
```

domluna, 2018-04-30:
can you share your code?

ywiyogo, 2018-04-30:
The code is in the Udacity [solution in Lesson 7, section 9](https://view6ddbca99.udacity-student-workspaces.com/notebooks/A-Star-City-Solution.ipynb)

domluna, 2018-07-13:
Added a note to the notebook related to the form the path should be in so it plots correctly.

closed 2018-07-13 by domluna