udacity/fcnd-issue-reports #251: Add TODOs to excercises in Moving into 3D lesson

jtuckerdunn, 2018-04-19:
Students commented in lesson feedback that when excercises do not have TODOs it was unclear which part of the codes students have to be working on.

https://classroom.udacity.com/nanodegrees/nd787/parts/5aa0a956-4418-4a41-846f-cb7ea63349b3/modules/0c12632a-b59a-41c1-9694-2b3508f47ce7/lessons/6e2a0873-e549-4045-a877-394ef572f93e/concepts/29ae6cac-904e-443c-a67f-9cc82da52f52

AndyAtUdacity, 2018-05-01:
Voxel Map Exercise : done
Random Sampling Exercise: done
Probabilistic Roadmap exercise: done
Receding Horizon exercise: this is meant to be an open-ended notebook without any "right answer". The "todo" is given intentionally as a high level suggested approach.

I'll let @domluna decide if he wants to make modifications to this last exercise.

domluna, 2018-07-13:
Since it's open-ended, specific TODOs don't fit in the picture.

closed 2018-07-13 by domluna