udacity/fcnd-issue-reports #256: Add links to references throughout Real World Planning Lesson

jtuckerdunn, 2018-04-19:
lesson feedback commenst: 

The lessons are good. But they need a lot of details\, resources\, and readings to help understand in depth such complex topics.

details are missed through the lessons...

ryan-keenan, 2018-04-21:
This is very true, mostly Nick's fault for including a bunch of stuff that wasn't really required for the project. Would be great to get versions of these exercises made that could easily transfer to the project implementation but that's going to take some significant effort.

AndyAtUdacity, 2018-05-01:
I added some text to the Intro of this lesson that sets expectations around what the level of depth is going to be. Would still be nice to add links to additional content though!