udacity/fcnd-issue-reports #277: In L17.12 Dead Reckoning 2D, incorrect matrices used for conversion between body frame and inertial frame

xcarnd, 2018-04-30:
In the solution, `R(phi)` is used for conversion from inertial frame to body frame and `R(-phi)` for body frame to inertial frame, which is not consistent with the former parts in this term.

For example, in the control part of this term, we are using `R(phi)` for converting collective thrust which is always aligned with z-axis in body frame to inertial frame.

xcarnd, 2018-05-01:
The same for L17.13. There is also descriptive text about the rotation matrix in the jupyter notebook of L17.13, saying it (the rotation matrix) is used for transform from body frame to global frame.

> We also would require the function which will provide us with the rotation matrix allowing us to transform the state, measured in body frame, into the global frame (rotation_matrix is already provided).

Yet in the solution, the inverse of rotation matrix was used for that purpose instead.

VladimerUdacity, 2018-05-10:
While in general 3d case we will use the inverse matrix of 2d rotation along the one axis we can substitute matrix inversion by simply rotating in opposite direction. Thus -phi.

xcarnd, 2018-05-16:
@VladimerUdacity In my opinion, 2d case is similar to 3d case. in 2d case, phi is the angle between body frame's x-axis and global frame's x-axis.

If we want to transform some vector represented in global frame into body frame, we can either:
1. rotate the base of global frame by phi, making it align with body frame
2. rotate the vector by -phi around global frame's axis. This corresponds to do matrix-vector multiplication R(-phi) * v.

That's why I think R(-phi) is a transformation matrix from global to body, and R(phi) for body to global.

dyfc, 2018-06-02:
When we talk about rotation matrix it is always good to remember about source and target coordinate systems?
I see a problem in rotation matrices in ObjectInThreeD.rotation_matrix() and IMU.rotation_matrix(). In both cases generated matrix assumes rotation from body to global frame but should be doing opposite, i.e. rotation from global frame to body frame.
It is easy to fix by changing sin() signs in r_x, r_y, r_z matrices (or negate phi, theta angles) and  multiplication order of matrices to: r_x*r_y*r_z