udacity/fcnd-issue-reports #282: In L18.20 EKF exercise, sigma is not properly initialized. `predict` method is not properly implemented

xcarnd, 2018-05-02:
In this exercise, the initial value of `sigma` is initialized as:

>sigma_0 = np.matmul(np.identity(3), >np.array([angle_error,velocity_sigma,position_sigma]))

which results in a 1x3 vector. But `sigma` is the covariance matrix so it shall be a 3x3 matrix.

The implementation for `predict` is also incorrect:

>    def predict(self, u):
>
>      previous_covariance = self.sigma
>        mu_bar = self.g(u)
>        G_now  = self.g_prime()
>        sigma_bar = np.matmul(G_now,np.matmul(previous_covariance,np.transpose(G_now))) + self.r_t
>
>        self.mu_bar  = mu_bar 
>        self.sigma_bar = sigma_bar
>
>        return mu_bar, sigma_bar

`sigma_bar` is added with `self.r_t`, which is the sensor noise matrix (It is correct in the solution notebook btw).

dyfc, 2018-05-25:
Fixed sigma_0 issue by making the following changes:
1. In EKF-Student.py
```python
 sigma_0 = np.identity(3) * np.array([angle_error,velocity_sigma,position_sigma])
```
2. In StateSpaceDisplayFor2D.py
```python
state_space_display_predict():
   ...
   initial_state_space=multivariate_normal([y,y_dot], [[sigma_0[2,2],0.0],[0.0,sigma_0[1,1]]])
   ...
```

annerajb, 2018-07-01:
Can confirm had to change q_t and r_t

WillieMaddox, 2018-12-23:
I came here to submit a bug report and came across this post.  I just wasted an hour debugging the same issues as those listed above but it looks like these issues were resolved 7 months ago.  Please make sure the changes are getting pushed to current workspaces.  Thanks.