udacity/fcnd-issue-reports #291: Lec 17.14 missing a negative sign in the z component [0 0 g]^

fredchyan, 2018-05-05:


fredchyan, 2018-05-05:
[Uploading 17.14.png...]

AndyAtUdacity, 2018-05-09:
This isn't actually  an error. Z-axis points down so g should be positive.

closed 2018-05-09 by AndyAtUdacity

fredchyan, 2018-05-09:
@AndyAtUdacity  could you please take a look at my comments above? but regardless I think it's missing negative sign somewhere, because R [0 0 g] != [g sin theta; -g sin phi cos theta; -g cos phi cos theta]

AndyAtUdacity, 2018-05-10:
Great catch and thank you for following up. You are absolutely right! I was clearly forgetting how accelerometers work :)

I'm going to add a note below the video indicating that this should be a -g

dyfc, 2018-06-02:
@AndyAtUdacity Here we should be careful with which rotation matrix to use. it is not stated explicitly in the video but, in my opinion, R should be a matrix rotating from global frame to body frame:
[image]

With this matrix, result of multiplication R*[0 0 -g].T is consistent with equation in video (at 1:30)