udacity/fcnd-issue-reports #293: Control project needs an x64 build config on Windows.

fierval, 2018-05-07:
Simulator.vcxproj needs to have a functional x64 configuration, otherwise Windows 10 is not really supported. Solution here: https://github.com/fierval/FCND-Controls-CPP

adrnp, 2018-05-08:
Were you able to run the Win32 (x86) build configuration?  If not, what was the error you were getting with that?

fierval, 2018-05-08:
It built fine, but I could not run it due to an OpenGL error. The error may have gone away if I just rebooted - I saw it happen during x64 builds as well, and rebooting would fix it, so I'd say this is not a p1 bug, however, it's simple enough to provide an x64 config (see above) and since an overwhelming majority is running x64 systems these days, this should probably be the default. This is one of those annoying little things that make devs look silly in the eyes of a user, and since we are paying users it's probably worth fixing.

fierval, 2018-05-08:
Here is the error I got: `freeglut: fgInitGL2: fghGenBuffers is NULL`

adrnp, 2018-05-12:
changes have been incorporated for building with x64.

closed 2018-05-12 by adrnp