udacity/fcnd-issue-reports #299: Crazyflie backyard flyer integration problem

mehmetyldz87, 2018-05-09:
Sometimes mission is not completed with different  local_waypoints ( Such as ; local_waypoints = [cp + [1.5, 0.0, 0.5], cp + [1.5, 0.8, 0.8], cp + [0.0, 1.5, 0.6], cp + [0.0, 0.0, 0.3]] )

mehmetyldz87, 2018-05-11:
This flight's Tlog file is here,

https://udacity-flying-car.slack.com/files/U98R8U6CC/FAM1GHZ2P/tlog.txt

mehmetyldz87, 2018-05-20:
Could you share a drone-integration video with us ? Does your drone behave like that ? Because  it seems , quad does not fly the box mission.  For this mission (local_waypoints = [cp + [1.0, 0.0, 0.5], cp + [1.0, 1.0, 0.5], cp + [0.0, 1.0, 0.5], cp + [0.0, 0.0, 0.5]]) , for example ,lets check first point , the quad is not going 1 meter on the x axis.  I wonder that I am doing something wrong in my code ( I use Drone Integration code which is given in Drone Integration Part).

Video Here

[image]

adrnp, 2018-05-24:
Here is a log file from my flight using the 1 meter box.  I will post a video to the slack drone integration channel.

[TLog-bf-crazyflie.txt](https://github.com/udacity/fcnd-issue-reports/files/2036363/TLog-bf-crazyflie.txt)

amiltonwong, 2018-05-25:
Hi, @adrnp ,

Could you share the code (backyard flyer for Crazyflies) here?

THX!

mehmetyldz87, 2018-07-22:
Is there any process? . On Slack Channel (#drone-integration), @AnastasiaK found a solution . This solution is about changing the crazyflie_connection.py file. 

Here 
```py
Index: crazyflie_connection.py
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- crazyflie_connection.py	(date 1530780548000)
+++ crazyflie_connection.py	(date 1530780548000)
@@ -179,6 +179,9 @@
         except AttributeError:
             print('Could not add state log config, bad configuration.')
 
+        # first step: reset the estimator to make sure all is good
+        self._reset_position_estimator()
+
         # start the write thread now that the connection is open
         self._running = True
         self._write_handle.start()
@@ -508,8 +511,6 @@
             e: current east position in meters
             altitde: desired altitude
         """
-        # first step: reset the estimator to make sure all is good
-        self._reset_position_estimator()
 
         # TODO: add to queue a command with 0 x,y vel, 0 yawrate, and the desired height off the ground
         cmd = CrazyflieCommand(CrazyflieCommand.CMD_TYPE_HOVER, (0.0, 0.0, 0.0, d)) 

```

mehmetyldz87, 2018-08-29:
Any process ?