udacity/fcnd-issue-reports #325: remove 'more videos' popup by youtube

asimonov, 2018-05-20:
something has changed around embedding of youtube videos in the lectures. 

now when video is paused there is extremely annoying popup ‘more videos’ by youtube. it is very very inconvenient for people who pause video to take notes/screenshots. it just drives me nuts. 

apparently there is a differing way to embed videos so this does not happen: https://liquidspark.com/removing-related-videos-youtube-embedded-video/

walkerp07, 2018-10-18:
I concur. How do we stop this?

flintt, 2018-10-21:
have the same problem, it's annoy

Xanthippe, 2018-10-21:
This is happening for me now as well - it's very inconvenient when pausing the video to screenshot it and see the subtitles. Once you close the pop-up once, it doesn't happen again for that video, but on Udacity every new page is a new video, so you can get 20 or 30 of these in a single lecture. If there's a way to disable it that would be great!
It takes up about a third of the video display in my usual set-up:
[image]

asimonov, 2018-10-21:
Yes, it is extremely annoying

_____________________________
Kind Regards,
Alexey Simonov

flintt, 2018-10-22:
adding two rules in uBlock chrome extension

> ##.ytp-pause-overlay
> www.youtube.com##.ytp-scroll-min.ytp-pause-overlay

flintt, 2018-10-22:
add two rules in uBlock (chrome extension)

> ##.ytp-pause-overlay
> www.youtube.com##.ytp-scroll-min.ytp-pause-overlay

Xanthippe, 2018-10-22:
@flintt This worked great, thanks.

asimonov, 2018-10-23:
worked for me too. thanks!

gaurkuber, 2018-10-23:
Awesome ! that helped. Thanks !!!

aalaa2045, 2018-10-23:
big thanks !!!!!!!!!!!!!!

foxxyz, 2018-10-24:
Thank you @flintt . This is ridiculous though...

Udacity, how has this intrusive feature slipped past? Please turn this off - is it literally getting in the way of education.

hshrews, 2018-10-24:
Thanks @flintt ! Unfortunately, though, blocking that popup banner doesn't fix the issue with closed caption. If I want to be able to read the CC with the video paused, I can't block the banner, so I'm back to manually closing it for each video. 

Really disliking Udacity more and more...

nzee, 2019-01-09:
Maybe it's time for Udacity to move away from Youtube.

Narthaxx, 2020-12-11:
> 
> 
> add two rules in uBlock (chrome extension)
> 
> > ##.ytp-pause-overlay
> > [www.youtube.com##.ytp-scroll-min.ytp-pause-overlay](http://www.youtube.com##.ytp-scroll-min.ytp-pause-overlay)

2 years later and I still love you for not just saying "Use ublock" and then insulting anyone who asked for more specifics (zapper/ picker _does not_ work on this)

lenainjaune, 2021-10-03:
@flintt Thank you ! You saved my nerves :+1: