udacity/fcnd-issue-reports #356: module named 'utils is not in the folder

tjphoton, 2018-06-26:
Lesson 21 GPS denied navigation lecture 17. Sensor Fusion Exercise:

```
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-abff0953a57e> in <module>()
      5 import scipy.stats as stats
      6 from bresenham import bresenham
----> 7 from utils import create_grid, inbounds

ModuleNotFoundError: No module named 'utils'
'''

annerajb, 2018-07-04:
Yup same issue. (also below it references the file path data/colliders.csv but the path should be colliders.csv

mvirgo, 2019-06-06:
Took a year apparently, but have added this file in now.

closed 2019-06-06 by mvirgo