udacity/fcnd-issue-reports #392: As of v1.1 the drone spawns inside a building and collides on most every activity

arush, 2020-03-24:
# Issue Template 

* Use tags in an effort to as specific as possible.
* If the issue is related to the simulator it's helpful to include log files described [here](https://docs.unity3d.com/Manual/LogFiles.html). The **Player** log is likely the most relevant.

I posted this in the [Q&A](https://knowledge.udacity.com/questions/87322) but received an unsatisfactory answer. Another student has posted the same question and also received an unsatisfactory answer from the same mentor. It seems we are being ignored now so am seeking help here.

How do we change the spawn location so it is not inside a building?

Thanks

pomiradek, 2020-04-11:
I have the same problem, is there any update ahead from @udacity?

sunsided, 2020-04-29:
Sadly I ended up in https://github.com/udacity/FCND-Simulator-Releases/issues/11#issuecomment-621002340 first, so I'm going to cross-post.

I'm able to "randomly" toggle between two broken states by using `SHIFT-R`.

The first one is empty:

[image]

The second one has that long building right on top of the starting position. Attempting to start the drone will immediately make it collide with the environment, breaking everything (even with `backyard_flyer_solution.py` of the planner).

[image]

I mean, there _is_ some beauty to it, it's just not particularly helpful.

[image]

midnight-dev, 2020-05-12:
@arush , @sunsided , @pomiradek I just happened to come upon this thread while browsing something completely unrelated. I don't have a fix, unfortunately, but it's clear that @udacity isn't planning to address any issue, be it new or old. There's not much that can be done short of asking around if anyone could contribute a fix or suggest a workaround. 

I'm a bit disappointed that the code is not available to be browsed on GitHub, but perhaps the source is available through releases. If so, someone could put up a fork if they feel like maintaining this. 

Seems the last maintainer was @domluna who is still very active. Might be worth a ping. Otherwise, the project may remain abandoned. At the very least, they would know who to contact regarding maintenance.

Best of luck to you all. Cheers o/

ilyko96, 2020-07-04:
I have the same issue. Windows 10. Simulator v0.1.1 from Mar 18, 2018.

Hesham-H98, 2020-09-18:
Been dealing with the same issue, so here is what I got..
I navigated through different releases of this simulator so as to find one without any problems, and this one [v0.1.0](https://github.com/udacity/FCND-Simulator-Releases/releases/tag/v0.1.0-motion-planning) gives not very satisfying result, but at least helps to keep you working on project properly..
The thing is that the drone will fly passing through the building as if it isn't there and continue its path
[image]
Still though it won't land on ground but on the building, but you can change the goal position anyway and land anywhere else!

vitalii-bulyzhyn, 2021-04-03:
Any updates on this issue?

cptanalatriste, 2022-04-26:
Facing the same issue, on Ubuntu 20 (64 bits). No updates from https://github.com/udacity?

JeffRubin, 2022-06-01:
Hesham-H98's comment is quite helpful.  It's a good workaround.  Note that Hesham-H98's comment refers to a version of Motion Planning, it's not the Simulator; the simulator seems to have come after Motion Planning and incorporated Motion Planning along with other tools, e.g. Backyard Flyer into one application.

An alternative workaround to Hesham-H98's approach is, with obstacles present, manually drive the vehicle through the mysterious building (and collide with it) and then manually move the vehicle to another location on the map where there is no obstacle. Then run the path planning algorithms from there. This has the added benefit of proving that the path planning algorithms can work from anywhere on the map, not just from the initial spawn point.  It's a messy workaround but it is pretty straightforward.

Much of this discussion was also captured in my question to the mentors (where they pointed me here): https://knowledge.udacity.com/questions/846330